kronometrix/recording #76: windows services running, no krd data available

sparvu, 2015-11-08:
We can see on some systems that Kronometrix services are up and running but no krd data is written to disk.

ohtu123, 2015-11-08:
sysrec.log records a drift time of 5000+ seconds. This means the system was sleeping or hibernating and recorders were running, keep ticking. 

```
2015-11-08 18:19:33 warning: drift detected, drift=61(s), interval=60(s)
2015-11-08 18:33:34 warning: drift detected, drift=842(s), interval=60(s)
2015-11-08 18:34:29 warning: drift detected, drift=56(s), interval=60(s)
2015-11-08 18:36:34 warning: drift detected, drift=61(s), interval=60(s)
2015-11-08 18:37:32 warning: drift detected, drift=59(s), interval=60(s)
2015-11-08 20:07:31 warning: drift detected, drift=5398(s), interval=60(s)
```

ohtu123, 2015-11-08:
as well other recorders show, hdwrec.log

```
2015-11-06 00:55:17 warning: drift detected, drift=61(s), interval=60(s)
2015-11-06 00:57:15 warning: drift detected, drift=59(s), interval=60(s)
2015-11-06 00:59:17 warning: drift detected, drift=61(s), interval=60(s)
2015-11-06 01:01:15 warning: drift detected, drift=59(s), interval=60(s)
2015-11-06 01:02:17 warning: drift detected, drift=61(s), interval=60(s)
2015-11-06 01:07:15 warning: drift detected, drift=59(s), interval=60(s)
2015-11-06 01:08:17 warning: drift detected, drift=61(s), interval=60(s)
2015-11-06 07:15:26 warning: drift detected, drift=21670(s), interval=60(s)
```

sparvu, 2015-11-08:
As well it seems we can not stop, restart the service after this thing is happening.

ohtu123, 2015-11-09:
Here data from sysrec:

```
2015-11-09 11:27:37 info: started
2015-11-09 13:54:18 warning: drift detected, drift=2859(s), interval=60(s), tnow=1447070058(epoch), sleep=53807(ms)
2015-11-09 13:55:13 warning: drift detected, drift=56(s), interval=60(s)
2015-11-09 13:56:18 warning: drift detected, drift=61(s), interval=60(s), tnow=1447070178(epoch), sleep=56942(ms)
2015-11-09 14:19:52 warning: drift detected, drift=635(s), interval=60(s), tnow=1447071592(epoch), sleep=56146(ms)
2015-11-09 14:20:49 warning: drift detected, drift=58(s), interval=60(s)
2015-11-09 14:21:52 warning: drift detected, drift=61(s), interval=60(s), tnow=1447071712(epoch), sleep=57098(ms)
2015-11-09 16:00:18 warning: drift detected, drift=5187(s), interval=60(s), tnow=1447077618(epoch), sleep=53698(ms)
2015-11-09 16:01:13 warning: drift detected, drift=56(s), interval=60(s)
```

immediately as the machine was online (from sleep, suspend to ram) the sysrec published and sender delivered data. This works fine for a normal situation. Im not sure whats happening if we run out of batteries and maybe the recorders/transporter do not function properly.

sparvu, 2015-11-09:
[image]

we can see clearly from above, the sysrec starting as soon as putting back online the machine. Putting back means hitting the power button once and waking up. Not booting it.

sparvu, 2016-01-05:
This is a problem within the drift system adjustment. This has been fixed in latest 1.0.18 RC1 release.

closed 2016-01-05 by sparvu