kronometrix/recording #95: cpurec Windows service stopped by itself

irimiab, 2016-02-08:
The cpurec service stopped by itself at 00:00 during (probably) log file rotation.
This is the last part of the file ```cpurec.log```:
```
2016-02-05 00:00:00 info: cktime - end log rotation, files rotated: 1
2016-02-05 14:25:00 info: stopped
2016-02-05 14:26:17 info: started
2016-02-06 00:00:00 info: cktime - start log rotation
2016-02-06 00:00:00 info: cktime - end log rotation, files rotated: 1
2016-02-08 11:31:14 info: started
```

At 14:26, on 5th, the latest version of the Kronometrix recording application has been updated. On 6th, at 00:00 the service stopped by itself. Today, on 8th, I manually started it (from the "Services" panel).

OS: Windows 10

sparvu, 2016-02-08:
Was your box up and running all this time ?

sparvu, 2016-02-08:
There are issues with kdr on laptops, sometimes some recorders do not run after running out of battery ...

irimiab, 2016-02-08:
I don't have a laptop, and yes, it was on all the time. All other recorders worked correctly.

sparvu, 2016-02-08:
That's strange. Can you please check:

   * Have any cpurec  raw data on daily ?

   * What was the last raw data for 5th ?

sparvu, 2016-02-08:
And can you try this update please http://www.kronometrix.org/pkgs/win/kdr-1.0.19rc5-win-x64.exe

irimiab, 2016-02-08:
No raw data in daily for 6th. On 5th, the last records in ```cpurec.krd``` are:
```
1454709540:cpu2:0.29:0.23:0.00:0.00:98.84:0.52
1454709540:cpu3:0.08:0.03:0.00:0.00:99.54:0.10
1454709540:cpu4:0.36:0.26:0.00:0.00:98.54:0.63
1454709540:cpu5:0.10:0.03:0.00:0.00:99.55:0.13
1454709540:cpu6:1.33:0.65:0.00:0.00:97.55:1.98
1454709540:cpu7:0.10:0.10:0.00:0.03:98.24:0.21
```

I will install that release soon.

sparvu, 2016-02-08:
That's 23:59:00 for cpurec raw data. Thanks for info. It should have worked. Curious can you send me your 5th of Feb daily dir by email a pls ?

sparvu, 2016-02-10:
Is this happening again  ?

irimiab, 2016-02-10:
Not until now.

sparvu, 2016-02-10:
IT was a re-spin, for RC5. Please once more, just we have same binary package all of us:
 
  *  http://www.kronometrix.org/pkgs/win/kdr-1.0.19rc5-win-x64.exe

irimiab, 2016-02-11:
So do I need to update the recorders? Didn't I just do that?

sparvu, 2016-02-11:
do it again pls. We are testing as well with Windows 10. No issues there in data .json restoration. Something is broken in your config.

sparvu, 2016-02-15:
This is the problem: could be a bug in delta values data from perfrawdata inside kernel:

```
The description for Event ID 0 from source Application cannot be found. Either the component that raises this event is not installed on your local computer or the installation is corrupted. You can install or repair the component on the local computer.

If the event originated on another computer, the display information had to be saved with the event.

The following information was included with the event: 

Illegal division by zero at cpurec line 308.
```

sparvu, 2016-02-15:
Please install RC5c which has a fix for bogus cpu data from win32 kernel. Please:

 * Download http://www.kronometrix.org/pkgs/win/kdr-1.0.19rc5c-win-x64.exe

 * C:\Program Files\Kronometrix\uninstall.exe -silent

 * C:\Users\xxx\kdr-1.0.19rc5c-win-x64.exe -silent

sparvu, 2016-02-15:
Before you install rc5c simple delete all log files from c:\ProgramData\Kronometrix\log\*.log

irimiab, 2016-02-15:
Could not download the file: 404. Please check the URL. Thanks!

sparvu, 2016-02-15:
ok, cool. Im actually reloading the cpurec with more debug info. in 10mins.

sparvu, 2016-02-15:
Try now please.

irimiab, 2016-02-15:
Ok, installed.

sparvu, 2016-02-22:
cupric, diskrec all have been coded to take care of illegal division by zero, in case bogus data comes in. Closing now, pls re-open if you see the troubles again.

closed 2016-02-22 by sparvu