kronometrix/recording #103: netrec metrics

DrQz, 2016-04-11:
KDR **netrec** fields currently look like this:
```
> redrooz@earth:~$ /opt/kronometrix/bin/netrec -h
> USAGE: netrec [-hV] | [interval [count]]
>  e.g. netrec 5       print continuously, every 5 seconds
>       netrec 1 5     print 5 times, every 1 second
> 
>  FIELDS:
>   #01 timestamp  : seconds since Epoch, time
>   #02 active     : TCP active connections, number
>   #03 passive    : TCP passive connections, number
>   #04 failed     : TCP failed connection attempts, number
>   #05 recv_rst   : TCP connection resets received, number
>   #06 est        : TCP connections established, number
>   #07 sent_rst   : TCP resets sent, number
>   #08 upks_recv  : UDP packets received, number
>   #09 upks_sent  : UDP packets sent, number
>   #10 fast_retr  : fast retransmits, number
>   #11 fwd_retr   : forward retransmits, number
>   #12 slow_retr  : retransmits in slow start, number
```
This seems woefully insufficient. It should contain snd/rcv and errors from all the ISO levels available in the **netstat -s** command (ICMP, TCP, IP, etc.), at least:
```
redrooz@earth:~$ netstat -s
Ip:
    1017388057 total packets received
    0 forwarded
    0 incoming packets discarded
    1016756023 incoming packets delivered
    906246357 requests sent out
    4 dropped because of missing route
Icmp:
    1101182 ICMP messages received
    659691 input ICMP message failed.
    ICMP input histogram:
        destination unreachable: 1101172
        echo replies: 10
    162 ICMP messages sent
    0 ICMP messages failed
    ICMP output histogram:
        destination unreachable: 42
        echo request: 120
IcmpMsg:
        InType0: 10
        InType3: 1101172
        OutType3: 42
        OutType8: 120
Tcp:
    3637262 active connections openings
    318 passive connection openings
    1800039 failed connection attempts
    45 connection resets received
    2 connections established
    1013614523 segments received
    1477898340 segments send out
    964450 segments retransmited
    1 bad segments received.
    1500837 resets sent
Udp:
    1918494 packets received
    38 packets to unknown port received.
    0 packet receive errors
    1918614 packets sent
UdpLite:
TcpExt:
    397 invalid SYN cookies received
    9 ICMP packets dropped because they were out-of-window
    68461 TCP sockets finished time wait in fast timer
    3838534 delayed acks sent
    347 delayed acks further delayed because of locked socket
    Quick ack mode was activated 7184 times
    4811310 packets directly queued to recvmsg prequeue.
    4114119174 bytes directly in process context from backlog
    3268755522 bytes directly received in process context from prequeue
    447511935 packet headers predicted
    9800132 packets header predicted and directly queued to user
    16862289 acknowledgments not containing data payload received
    706174761 predicted acknowledgments
    16637 times recovered from packet loss by selective acknowledgements
    18 bad SACK blocks received
    Detected reordering 38 times using SACK
    Detected reordering 17 times using time stamp
    19 congestion windows fully recovered without slow start
    65 congestion windows partially recovered using Hoe heuristic
    3368 congestion windows recovered without slow start by DSACK
    61 congestion windows recovered without slow start after partial ack
    TCPLostRetransmit: 492
    3 timeouts after SACK recovery
    1 timeouts in loss state
    89956 fast retransmits
    5777 forward retransmits
    557 retransmits in slow start
    435680 other TCP timeouts
    TCPLossProbes: 406976
    TCPLossProbeRecovery: 245887
    21 SACK retransmits failed
    7194 DSACKs sent for old packets
    398332 DSACKs received
    36 DSACKs for out of order packets received
    10 connections reset due to unexpected data
    35 connections reset due to early user close
    1 connections aborted due to timeout
    TCPSACKDiscard: 9
    TCPDSACKIgnoredOld: 8
    TCPDSACKIgnoredNoUndo: 192079
    TCPSpuriousRTOs: 13
    TCPSackShifted: 206249
    TCPSackMerged: 89105
    TCPSackShiftFallback: 172606
    TCPRetransFail: 159
    TCPRcvCoalesce: 16536810
    TCPOFOQueue: 460634
    TCPChallengeACK: 7
    TCPSYNChallenge: 1
    TCPSpuriousRtxHostQueues: 2250
IpExt:
    InBcastPkts: 121784
    InOctets: 854093680
    OutOctets: -162930218
    InBcastOctets: 36094302
    InNoECTPkts: 1109844394
```
Moreover, these are just **counts**. It is generally more helpful is they are represented as **throughput** metrics (i.e, counts per sample interval) because that is more closely allied with the concept of network bandwidth.

DrQz, 2016-04-11:
**Example**
There is no change in the _netrec_ values:
```
...
1460338499:3638536:318:1801313:45:2:1502111:1918537:1918657:89984:5780:557
1460338509:3638536:318:1801313:45:2:1502111:1918537:1918657:89984:5780:557
1460338519:3638536:318:1801313:45:2:1502111:1918537:1918657:89984:5780:557
...
```
even when I _ping_ **Yahoo.com** ....
```
PING fd-fp3.wg1.b.yahoo.com (98.139.180.149) 56(84) bytes of data.
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=1 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=2 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=3 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=4 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=5 ttl=52 time=120 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=6 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=7 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=8 ttl=52 time=118 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=9 ttl=52 time=122 ms
64 bytes from ir1.fp.vip.bf1.yahoo.com (98.139.180.149): icmp_req=10 ttl=52 time=118 ms
```
because there is no **ICMP** level reported in _netrec_.

sparvu, 2016-04-11:
thanks. netrec needs a bit of re-work. it does not include many UDP, TCP stats and will require some polishing. As well majority of netrec parameters are counters.

sparvu, 2017-07-12:
this will be handled after we have a basic snmprec recording in. snmp recorder planned for Q3 2017.

DrQz, 2017-07-12:
👍