kronometrix/recording #114: diskrec debian tmp size is not calculated

taropalo, 2016-09-01:
#df -h
/dev/mapper/kx14client--vg-tmp   268M   68M  182M  27% /tmp

#diskrec 
1472715953:dm3tmp:0.00:0.00:0.00:0.00:0.00:0.00:1:1:0

Even after using /tmp 68% diskrec doesn't have values. Same goes with /var

root@kx14client:~# df -h
Filesystem                       Size  Used Avail Use% Mounted on
/dev/dm-0                        2.8G  780M  1.8G  30% /
/dev/vda1                        236M   33M  191M  15% /boot
/dev/mapper/kx14client--vg-home  2.8G  1.1G  1.6G  40% /home
/dev/mapper/kx14client--vg-tmp   268M  168M   82M  68% /tmp
/dev/mapper/kx14client--vg-var   1.4G  184M  1.1G  15% /var

root@kx14client:~# /opt/kronometrix/bin/diskrec 
1472716187:dm0root:0.00:0.00:0.00:0.00:0.00:0.00:3:2:34
1472716187:dm2var:0.00:0.00:0.00:0.00:0.00:0.00:2:2:0
1472716187:dm3tmp:0.00:0.00:0.00:0.00:0.00:0.00:1:1:0
1472716187:dm4home:0.00:0.00:0.00:0.00:0.00:0.00:3:2:34
1472716187:vda:0.00:0.00:0.00:0.00:0.00:0.00:NA:NA:NA
1472716187:vda1boot:0.00:0.00:0.00:0.00:0.00:0.00:1:1:0
1472716187:vda2:0.00:0.00:0.00:0.00:0.00:0.00:NA:NA:NA
1472716187:vda5:0.00:0.00:0.00:0.00:0.00:0.00:NA:NA:NA

sparvu, 2016-09-01:
fixed in latest. test again now

sparvu, 2016-09-01:
should work fine. reopen pls if you still see this again.

closed 2016-09-01 by sparvu