kronometrix/recording #150: iaq transporter machine-id

sparvu, 2017-06-05:
IAQ transporters again have change their ID. Here:

```
root@strato3:/opt/kronometrix/etc# date
Mon  5 Jun 21:09:37 EEST 2017

root@strato3:/opt/kronometrix/etc# ls -lrt /etc/machine-id 
-rw-r--r-- 1 root root 33 Jan  1  1970 /etc/machine-id
```

This is somehow strange.

sparvu, 2017-06-05:
```
root@strato3:/home/pi# cat /etc/machine-id 
8f6b910d98cf49448ae9ebe23970da41
```

and after reboot:

```
pi@strato3:~ $ cat /etc/machine-id 
919fc0acf2b84c6f8369ab51909904a8
```

sparvu, 2017-06-05:
This is:

```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@strato3:~ $ uname -a
Linux strato3 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux
```

sparvu, 2017-06-05:
problem seems /etc/machine-id is a tmpfs :) wtf is wrong with these guys ? 

```
pi@strato3:~ $ cat /etc/mtab | grep machine
tmpfs /etc/machine-id tmpfs ro,mode=755 0 0
```

sparvu, 2017-06-05:
and this under another RBPI, totally different. WTF again. 

```
krmx@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux
krmx@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
krmx@raspberrypi:~ $ 
krmx@raspberrypi:~ $ ls -lrt /etc/machine-id 
-r--r--r-- 1 root root 33 May 27  2016 /etc/machine-id
krmx@raspberrypi:~ $ 
krmx@raspberrypi:~ $ grep machine /etc/mtab
krmx@raspberrypi:~ $ 
```

This is a mess.

taropalo, 2017-06-12:
It must be some bug in Rasbian. I cannot see why /etc/machine-id should be mounted as tmpfs. Now /var/lib/dbus/machine-id is created and that is now constant between boots. So it should take care of this problem.