kronometrix/recording #157: More KDR info

DrQz, 2017-08-23:
This is not just an engineering repo but a **sales opportunity**. 

You should add links to web pages that explain more about KDR, show dashboard demos, and lead into all your marketing collateral.

sparvu, 2017-08-23:
not sure if this (kronometrix/recording) is the right place. As soon as I put up kronometrix.org and kronometrix.com sites there will be proper places where people can check and see things. Right ?

DrQz, 2017-08-23:
If you want to increase sales, just do it.

sparvu, 2017-08-28:
of course we are trying to make everything for sales under kronometrix.com ...

DrQz, 2017-08-28:
This github README contains plenty of tech info about what gets recorded, links to download,  registration and a "sales" email address. That's all fine.

But, what if someone starts on this page (which has happened) and they would like to read more background about your KDR product(s) before taking the plunge. Why not provide a link or two to pages that already exist and provide them with an overview? Like [this page](http://www.kronometrix.com/products/)?  Very simple to add.

Ppl will not sign up or register for what they don't really understand. Therefore, give them an easy opportunity to find out more. I know that's not usually done from a README but, so what!?