kronometrix/recording #165: header information when run in non logging mode

sparvu, 2017-12-06:
All data recorders must present a header information, describing each metric when run interactively. In logging mode, the header must not be present since data can be used only by machines.

sparvu, 2018-05-05:
we need to nicely print headers in interactive mode. still to be done for 1.3.x release

sparvu, 2018-05-05:
next major release will be 1.3.8

DrQz, 2018-05-07:
**New Ontology Groupings for Fields**

* Rename some fields, e.g., Processor or Cores (instead of CPU) - may want it short
* Current CPU field really contains O/S scheduler metrics, e.g., nicepct
* Make MEM fields a separate KDR record, e.g., mermen, while retaining essentials in 'sysrec'
* Many metrics (especially for passive resources) are **zeros** 
* Are all the O/S metrics in sysrec really **necessary**?

Current KDR metrics organization:
```
FIELDS:
  CPU
   #01 timestamp  : seconds since Epoch, time
   #02 cpupct     : cpu utilization, across all cpus, number
   #03 sumpct     : sum of all cpus utilization, number
   #04 headpct    : headroom cpu available, all cpus, number
   #05 userpct    : cpu utilization, user space in percent, number
   #06 nicepct    : cpu utilization, user space with nice priority, number
   #07 sysct      : cpu utilization, system space, number
   #08 idlepct    : cpu utilization, idle state, number
   #09 iowaitcpt  : cpu percentage in idle state because an I/O operation 
                     is waiting to complete, number
   #10 irqpct     : cpu percentage servicing interrupts, number
   #11 softirqpct : cpu percentage servicing softirqs, number
   #12 stealpct   : cpu percentage of time spent in other operating systems
                    when running in a virtualized environment, number
   #13 runqsz     : run queue length, tasks waiting for run time, number
   #14 plistsz    : tasks in the task list, number
   
```

A better ontological organization for "CPU" would be:

```
FIELDS:
  SAMPLE
   #01 timestamp  : seconds since Epoch, time

  CORES
   #02 cpupct     : cpu utilization, across all cpus, number    <--- %cpu AVERAGED over all cores
   #03 sumpct     : sum of all cpus utilization, number         <--- TOTAL %cpu can be > 100%
   #04 headpct    : headroom cpu available, all cpus, number
   #05 userpct    : cpu utilization, user space in percent, number <--- s.b. user TIME
   #07 sysct      : cpu utilization, system space, number          <--- s.b. syspct? and sys TIME
   #08 idlepct    : cpu utilization, idle state, number            <--- s.b. idle TIME
   #09 iowaitcpt  : cpu percentage in idle state because an I/O operation 
                     is waiting to complete, number
   #12 stealpct   : cpu percentage of time spent in other operating systems
                    when running in a virtualized environment, number
   
  SCHED
   #06 nicepct    : cpu utilization, user space with nice priority, number
   #10 irqpct     : cpu percentage servicing interrupts, number
   #11 softirqpct : cpu percentage servicing softirqs, number
   #13 runqsz     : run queue length, tasks waiting for run time, number
   #14 plistsz    : tasks in the task list, number
   #44 avg1       : LA of the last minute, number
   #45 avg5       : LA of the last 5 minutes, number
   #46 avg15      : LA of the last 15 minutes, number

```

DrQz, 2018-05-07:
[image]

DrQz, 2018-05-07:
I realized later that I prolly should've put this under a separate section (now done).

There's no real urgency about this stuff. But ultimately, I can make the case that it is:
* a very important issue for Krono product differentiation
* has direct bearing on Krono perf/data visualization tools

The above reorganization of sysrec metrics is merely a demonstration, not claiming it's final form.