kronometrix/recording #172: missing data on MODBUS RTU

sparvu, 2018-03-27:
I have noticed that rs485rec will not efficiently record data from +4 devices attached on a serial line using MODBUS RTU. This is the top view for a setup using GMW90 devices and latest KDR

[image]

For example this is how a 6hr period looks like:

[image]

We don't have access to the rs485rec logs but most likely here we are seeing: _cannot access device xxxx, check rs485rec.json_ or there is a transport issue. Looking closely for a certain device I can see that for example during certain periods of time, we are missing 1 minute data, like here:

[image]

[image]

thats 9:35AM missing data for all metrics. Checking against raw data set, for example for this device we can see clearly we are missing 9:35AM

```
"1522132415","23.99","25.72","3.14","570.00"
"1522132475","23.97","25.75","3.14","569.00"
 ????????????????????????????????????????
"1522132595","23.96","25.75","3.13","563.00"
"1522132655","23.95","25.77","3.14","570.00"
```

sparvu, 2018-03-30:
range operator will be defined as this. for example this is a Vaisala GMW90 device where we access all needed metrics via function 3, and using here range operator which provide the starting register address and the maximum length to read from. 

Using range a single call shall be issued to the device reading *all* values ... 

```
        {
          "unit": 3,
          "vendor": "vaisala",
          "model": "gmw90",
          "description": "Vaisala GMW90 Series",
          "function": 3,
          "range": {
              "register": 256,
              "length": 4
          },
          "data": [
            {
              "name": "ta",
              "type": "int16",
              "register": 258,
              "length": 1,
              "factor": 0.01
            },
            {
              "name": "rh",
              "type": "uint16",
              "register": 257,
              "length": 1,
              "factor": 0.01
            },
            {
              "name": "td",
              "type": "int16",
              "register": 259,
              "length": 1,
              "factor": 0.01
            },
            {
              "name": "co2",
              "type": "uint16",
              "register": 256,
              "length": 1,
              "factor": 1
            }
          ]
        }
```

sparvu, 2018-04-01:
Lets measure here, two cases, using a G01 device on MODBUS RTU. 
 
 * Case 1: reading individual metrics, using multiple calls, 1 connection opened  
```
One by one...
Value r1=1229.00
Value r2=6.10
Value r3=23.40
Value r4=23.40
single metric: 1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

One by one...
Value r1=1234.00
Value r2=6.10
Value r3=23.40
Value r4=23.40
single metric: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

One by one...
Value r1=1242.00
Value r2=6.20
Value r3=23.40
Value r4=23.30
single metric: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

One by one...
Value r1=1251.00
Value r2=6.20
Value r3=23.40
Value r4=23.20
single metric: 1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

```

 * Case 2: reading a range of metrics, using a single call, 1 connection opened 
```
Range call...
1229 62 234 234
range metrics: 1 wallclock secs ( 0.00 usr +  0.00 sys =  0.00 CPU)

Range call...
1234 61 234 234
range metrics: 1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

Range call...
1242 62 234 233
range metrics: 0 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

Range call...
1251 62 234 232
range metrics: 0 wallclock secs ( 0.00 usr +  0.00 sys =  0.00 CPU)
```

sparvu, 2018-04-01:
Partial conclusions:

  * there is no big difference between single (range) / multiple  calls 

  * range seems more efficient in terms of CPU usage 

  * overall there is no difference in how we call registers on G01 device

sparvu, 2018-04-01:
Testing MSD-1xxx device, another RTU device we have around for multiple calls. 

  * Running every 1,2 seconds, it works fine as long as no other recorder is using the line 

  * we can see around 2,3 seconds spent in fetching metric by metric from MSD device

  

```
Multiple calls...
Value r1=2.70
Value r2=4.20
Value r3=25.11
Value r4=30.24
Value r5=962.00
Value r6=0.22
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Multiple calls...
Value r1=2.80
Value r2=3.90
Value r3=25.11
Value r4=30.24
Value r5=963.00
Value r6=0.23
Time: 2 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

Multiple calls...
Value r1=2.80
Value r2=3.90
Value r3=25.11
Value r4=30.24
Value r5=962.00
Value r6=0.23
Time: 2 wallclock secs ( 0.00 usr +  0.01 sys =  0.01 CPU)

Multiple calls...
Value r1=2.90
Value r2=3.90
Value r3=25.11
Value r4=30.28
Value r5=963.00
Value r6=0.23
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Multiple calls...
Value r1=2.90
Value r2=4.30
Value r3=25.11
Value r4=30.28
Value r5=963.00
Value r6=0.23
Time: 3 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Multiple calls...
Value r1=2.80
Value r2=4.20
Value r3=25.11
Value r4=30.12
Value r5=963.00
Value r6=0.23
Time: 2 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

Multiple calls...
Value r1=2.80
Value r2=4.50
Value r3=25.12
Value r4=30.12
Value r5=963.00
Value r6=0.23
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Multiple calls...
Value r1=2.70
Value r2=3.90
Value r3=25.12
Value r4=30.00
Value r5=964.00
Value r6=0.23
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Multiple calls...
Value r1=2.70
Value r2=3.80
Value r3=25.12
Value r4=30.00
Value r5=964.00
Value r6=0.23
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)
```

sparvu, 2018-04-01:
and here the difference between single and multiple calls. nothing amazing. 

```
Multiple calls...
Value r1=5.60
Value r2=7.70
Value r3=25.15
Value r4=23.00
Value r5=589.00
Value r6=0.06
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Single call/range...
16563 13107 16630 26214 16841 7864 16824 0 17427 16384 15741 62390
time: 1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)

Multiple calls...
Value r1=5.70
Value r2=7.10
Value r3=25.14
Value r4=22.92
Value r5=584.00
Value r6=0.06
Time: 2 wallclock secs ( 0.02 usr +  0.00 sys =  0.02 CPU)

Single call/range...
16566 26214 16611 13107 16841 7864 16823 23593 17425 0 15741 62390
time: 1 wallclock secs ( 0.01 usr +  0.00 sys =  0.01 CPU)
```

sparvu, 2018-04-01:
partical conclusions:

  * G01, MSD-XXXX devices behave normally for single or multiple range addresses metrics 

  * the difference between single or multiple range addresses simple do not matter as proofed in this example 
  
  * still range operator might make a difference for different other devices or improve the CPU consumption

DTPopa, 2018-04-01:
Should we have a setting in the actual recorder / a recorder version where we can use this, just let us know, and we will use it.
The rationale for using a single interrogation to get a single answer containing multiple parameters, on a communication line, goes way beyond this particular case which is still a proof that this should be the way the recorder should work.

sparvu, 2018-04-01:
address range is under work and it will land soon on rs485rec. was there already from day 1 planned but never implemented. no worries about that.

now when it comes to this particular problem, reported here:

  * not sure whats the issue here 
  
  * slow readings from the device using wire ? 

  * slow reading from device not using wire ?

  * no guarantee that address range will improve anything, unless we know what we are dealing with

sparvu, 2018-04-02:
here another look after 24hours:

[image]

sparvu, 2018-04-02:
Another thing which I found was that we still had within rs485rec the sleep 2sec after each device. 

```
        for my $key (keys %devices) {

            my $value = $devices{$key};
            my ($rawfile, $dev_name, $modbus_id, $modbus_func, $desc) = split( /:/, $key );
            my @values = split( /:/, $value );
            eval {
                exec_modbus($client, $rawfile, $dev_name, $modbus_id, $modbus_func, @values);
            };

            if ($@) {
                if ($logging) {
                    write_log ("error: cannot access device $dev_name, check $cfile");
                } else {
                    print "error: cannot access device $dev_name, check $cfile\n";
                }
            }
            sleep 2;
        }
```

This can cost some time, on installations with +5 devices ... or slow devices ... Needs testing. For example here with and without the delay:

 * with the 2sec delay, 3 devices
```
1522652651:msd1618_1:24.71:33.04:7.36:889.00:0.87:1.70:3.80
1522652653:g01_2:23.70:27.20:3.69:861.00:10.90
1522652655:g01_3:23.20:25.10:2.13:993.00:6.90
        8.14 real         0.82 user         0.07 sys
```

 * without the 2sec delay, 3 devices
```
1522652697:msd1618_1:24.72:32.88:7.30:889.00:0.88:1.70:3.40
1522652698:g01_2:23.70:27.20:3.69:863.00:10.90
1522652699:g01_3:23.20:25.20:2.19:993.00:6.90
        5.44 real         0.83 user         0.06 sys
```

Not a big difference  for such small setup. Might differ on larger setups/

sparvu, 2018-05-05:
range filter has been implemented on latest git snapshot. closing now

closed 2018-05-05 by sparvu