kronometrix/recording #174: KDR Sampling Rate?

DrQz, 2018-05-04:
After all these years, where is the f*cking **default sample rate** documented in these web pages?

sparvu, 2018-05-05:
Fixed. 
https://github.com/kronometrix/recording/blob/master/README.md#the-recorder

KDR recorders don't really have a default sample rate value since they can be used interactive or logging, machine mode for continuously monitoring. But you are correct for continuously monitoring there is a default sample rate configured which can be overridden and I have documented now.

sparvu, 2018-05-05:
closing now

closed 2018-05-05 by sparvu

DrQz, 2018-05-05:
Not done yet..

* Data Recording @ **60 sec default** sampling interval also needs to be stated clearly on [kronometrix.ORG] page
* Ditto for corresponding [kronometrix.COM](http://www.kronometrix.com/products/data-recording-module-design/) page

If you're going to maintain 2 web sites, both need to be _consistent_ ... as well as consistent with [github](https://github.com/kronometrix/recording/blob/master/README.md#the-recorder).

sparvu, 2018-05-05:
* kronometrix.org - will be ramped down or improved. remains to be seen. dont have much time to keep it up. majority of things for .org are moved directly to github.com 

 * kronometrix.com - under work. yes. we will have a clear product section there for kronometrix data recording where I will document that.