kronometrix/recording #176: Ontological Groupings for sysrec Fields

DrQz, 2018-05-07:
* Rename some fields, e.g., Processor or Cores (instead of CPU) - may want it short
* Current CPU field really contains O/S scheduler metrics, e.g., nice pct
* Make MEM fields a separate KDR record, e.g., mermen, while retaining essentials in 'sysrec'
* Many metrics (especially for passive resources) are zeros
* Are all the O/S metrics in sysrec really necessary?

Current KDR metrics organization:
```
FIELDS:
  CPU
   #01 timestamp  : seconds since Epoch, time
   #02 cpupct     : cpu utilization, across all cpus, number
   #03 sumpct     : sum of all cpus utilization, number
   #04 headpct    : headroom cpu available, all cpus, number
   #05 userpct    : cpu utilization, user space in percent, number
   #06 nicepct    : cpu utilization, user space with nice priority, number
   #07 sysct      : cpu utilization, system space, number
   #08 idlepct    : cpu utilization, idle state, number
   #09 iowaitcpt  : cpu percentage in idle state because an I/O operation 
                     is waiting to complete, number
   #10 irqpct     : cpu percentage servicing interrupts, number
   #11 softirqpct : cpu percentage servicing softirqs, number
   #12 stealpct   : cpu percentage of time spent in other operating systems
                    when running in a virtualized environment, number
   #13 runqsz     : run queue length, tasks waiting for run time, number
   #14 plistsz    : tasks in the task list, number
```
   
A better ontological organization for "CPU" would be:

```
FIELDS:
  SAMPLE
   #01 timestamp  : seconds since Epoch, time

  CORES
   #02 cpupct     : cpu utilization, across all cpus, number    <--- %cpu AVERAGED over all cores
   #03 sumpct     : sum of all cpus utilization, number         <--- TOTAL %cpu can be > 100%
   #04 headpct    : headroom cpu available, all cpus, number
   #05 userpct    : cpu utilization, user space in percent, number <--- s.b. user TIME
   #07 sysct      : cpu utilization, system space, number          <--- s.b. syspct? and sys TIME
   #08 idlepct    : cpu utilization, idle state, number            <--- s.b. idle TIME
   #09 iowaitcpt  : cpu percentage in idle state because an I/O operation 
                     is waiting to complete, number
   #12 stealpct   : cpu percentage of time spent in other operating systems
                    when running in a virtualized environment, number
   
  SCHED
   #06 nicepct    : cpu utilization, user space with nice priority, number
   #10 irqpct     : cpu percentage servicing interrupts, number
   #11 softirqpct : cpu percentage servicing softirqs, number
   #13 runqsz     : run queue length, tasks waiting for run time, number
   #14 plistsz    : tasks in the task list, number
   #44 avg1       : LA of the last minute, number
   #45 avg5       : LA of the last 5 minutes, number
   #46 avg15      : LA of the last 15 minutes, number
```

DrQz, 2018-05-07:
There's no real urgency about this stuff, but ultimately, I can make the case that it is:
* a very important issue for Krono **product differentiation**
* has direct bearing on Krono perf/data **visualization tools**

The above reorganization of _sysrec_ metrics is merely a demonstration, not claiming it's final form. I'm still thinking about it.

We need more data examples from different types of workloads.

DrQz, 2018-05-07:
[image]

DrQz, 2018-05-07:
Between 'cpupct' and 'avg15' there are 45 _sysrec_ metrics. That corresponds to an N = 45-dimensional space. Eliminating _sysrec_ fields with either all **zeros** or all a **fixed value** (e.g., memory size)

MEM metrics are reduced to just:
* memusedpct 
* memused 
* cached 
* realfree 
* realfreepct 

DISK metrics are reduced to just:
* kbytes
* rwkbytes

NIC metrics are reduced to just:
* rxpkt  
* rxkbytes 
* txpkt 
* txkbytes 
* rxtxpkt 
* rxtxkbytes

sparvu, 2018-05-08:
if we change the sysrec message that means a change toawards the analytics system. I need to finish some urgent bugs for rs485rec - after that I will take a look.

Comments:

  * some metrics are 0 because of the workload or specific load on the machine 

  * of course we can rename anything we want, but I would not like to jump easily on re-grouping them since it does affect the message structure and I need to re-organize the data towards the analytics 

  * Storage part must offer throughput in MB|KB and IOPS 

  * Network part must offer as well throughput in Mbit and packets / sec