kronometrix/recording #178: New recorder for the AXIS camera's traffic app

irimiab, 2018-08-01:
We need  to integrate the traffic data from AXIS cameras into Kronometrix. The traffic data is available as JSON content at a certain URL (with basic authentication).

One idea was to extend `httprec` to make requests to the cameras the same way it does with status pages for the web servers. But there are some differences:
- first of all, `httprec` seemed to be a "webserver status recorder" and extending it with the traffic data is a bit strange
- in its current form, `httprec` doesn't know basic authentication, and we need this for the traffic data
- also, `httprec` doesn't parse JSON content, but only basic text (using regular expressions)

So another idea is to implement a new recorder `trafficrec`, or `axistrafficrec`, that does the same thing as `httprec`, but supports basic authentication and JSON parsing and is crafted specifically for the traffic data.

So what do you think? Should we extend `httprec` or write a new recorder for the traffic data?
Could you assist me in adding these 2 features (basic auth and JSON parsing)?

sparvu, 2018-08-03:
Sorry, I was extremely busy moving things around between different addresses. Now Im coming back to life.

Comments:

 * I already had a look some weeks back on this issue. The AXIS camera is nothing more or less than a yet another HTTP server which can offer different type of information

 * some basic information comes from the HTTP server itself AXIS vendor 

 * some additional information is served by the same HTTP server but provided by some 3rd parties apps deployed on the camera 

  * thats why we have httprec to speak to different HTTP servers and fetch different messages. Nobody said the httprec is for Apache or NGINX only

  * and very good if it does require authentication httprec must be enhanced to do that. there are as well dozens of cases where httprec will require as well that for ICT

From this aspect nothing is different than a TomCat. Yes, it can be easily added JSON parsing (like a breeze in Perl) and authentication (probable requires some testing)

So, I just need to know here the following:

 * a basic message (something simple returned by the camera itself) such way like a status, bytes transferred 

 * the real traffic message (which will be your interest)

Provide me a real test case where this camera be accessed, send me the details offline. I will start to work on this and enhance httprec to be able to fetch these messages from AXIS.

irimiab, 2018-08-06:
I will add the necessary details today: how to get the data from our system (the camera installed at Iasi) and also how should the message (for Kronometrix) look like. But this is already described in the "doc" file sent by Darius.
Anyway, I will synthesize it here.

sparvu, 2018-08-06:
Yes. good. Update the CR. For test and verifications add a simple check message for this camera that I could test first the HTTP from camera itself.

sparvu, 2018-08-06:
I explained many times. Extra docs are good, important for brain storming etc. But we need CRs in the defect mgmt.

irimiab, 2018-08-06:
I don't know what you mean by "check" message. I will describe how you can acquire data from the camera.

sparvu, 2018-08-06:
well what I meant is this:

  * do you want to have a set of basic metrics, very few from the camera itself which should be more or less reporting on the status of the camera: bytes transferred, number or errors, etc . this can be a simple message which should always report and check on the status of the camera, where no other apps or additional apps are used within camera 
  
  * additional your traffic information can be another message where the information will come from camera and app deployed on it

irimiab, 2018-08-06:
Oh, I see. No, this issue is only for the traffic information. In fact, we should keep these separated because they have different purposes and usage. So we will start only with the traffic information.

sparvu, 2018-08-06:
in general this is a method of checking how things are functioning on the bare metal itself with no additional things deployed. For example if the app might fail, does the camera still work ? why it has failed ? etc ... 

That's why in general when we have to pull data from things which employ a generic base or platform we first check the base and then the deployed applications etc ... 

but you know what u want.

sparvu, 2018-08-06:
no worries. just craft what u want. you most likely will return to that a bit later.

irimiab, 2018-08-06:
So the basic concept for traffic on AXIS cameras is this: you can configure a "probe" which is a zone on the image, on which the application does the traffic numbering. Generally, you set a probe on each line of the road.
So in Kronometrix, one probe will be one device. Probes are identified by numbers, so I propose we name our devices "probeX", where "X" is the number of the probe; e.g. "probe0", "probe1"

The recorder must be configured with the number of probes to query. So in `httprec.json` we should probably have these settings for the traffic thing:
- url
- username
- password
- number of probes

For each probe, the recorder should do a request to this URL (with basic auth):
```
http://<url>/local/VehicleCounter/getStatsPerClass.cgi?interval=60&pid=<probeID>
```
where `<url>` is the URL of the camera and `<probeID>` is a number from 0 to `number_of_probes-1` (probes are 0-based indexed).

Example of a request:
```
http://109.166.254.217/local/VehicleCounter/getStatsPerClass.cgi?interval=60&pid=0
```

The response:
```json

{
	"id": 6480797,
	"build_version": "v3.4.13",
	"cameraId": "accc8e62e39d",
	"session_id": "accc8e62e39d:20180620:204305",
	"gps_lat": "43.788286",
	"gps_lon": "11.226857",
	"query": "getStatsPerClass",
	"localtime": "2018-08-06T09:02:45",
	"interval": 60,
	"probe": 0,
	"count": [
		{
			"cls": 2,
			"count()": 2,
			"min(len)": 420,
			"avg(len)": 432.0,
			"max(len)": 444,
			"min(spd)": 10,
			"avg(spd)": 15.5,
			"max(spd)": 21
		},
		{
			"cls": 3,
			"count()": 1,
			"min(len)": 456,
			"avg(len)": 456.0,
			"max(len)": 456,
			"min(spd)": 51,
			"avg(spd)": 51.0,
			"max(spd)": 51
		},
		{
			"cls": 4,
			"count()": 7,
			"min(len)": 608,
			"avg(len)": 686.285714285714,
			"max(len)": 784,
			"min(spd)": 48,
			"avg(spd)": 66.8571428571429,
			"max(spd)": 90
		},
		{
			"cls": 5,
			"count()": 5,
			"min(len)": 896,
			"avg(len)": 1327.4,
			"max(len)": 1834,
			"min(spd)": 57,
			"avg(spd)": 89.0,
			"max(spd)": 156
		}
	]
}
```

From this response, the section `count` is of interest.
This section contains maximum 5 blocks, each block being for a class of vehicles. The vehicles are grouped in 5 size classes; if during the last minute there was no vehicle in a certain class, it will be missing from the `count` section altogether. 

From each vehicles class we will need the fields `count()` and `avg(len)`. If the class is missing, these values will be 0.

We will have a Kronometrix message like this:
```
<timestamp>:<probeID>:<count_cls0>:<avg_len_cls0>:<count_cls1>:<avg_len_cls1>...<count_cls5>:<avg_len_cls5>
```

For example, for the response above, the message will be:
```
<timestamp>:probe0:0:0.0:0:0.0:2:432.0:1:456.0:7:686.28:5:1327.4
```

A similar request will be made for probe 1 too.
For the URL in the example, I will send the credentials on SMS  📱

The recorder should run **every minute** because the parameter `interval=60` in the request URL means we are querying for data for the last 60 seconds.

sparvu, 2018-08-06:
will review now.

sparvu, 2018-08-06:
so the number of classes is always from 1 to 5 ? And some classes might not be present at all ? Correct ?

irimiab, 2018-08-06:
Yes, the number of classes is always 1 to 5, some might miss some during some minutes, but might reappear afterwards. It depends on what cars pass on the road.

sparvu, 2018-08-06:
ok. I think, I got the point. I am committing now the UPS changes to KDR and then I will start implementing the new httprec data message for AXIS. Will ask more as I go.

sparvu, 2018-09-03:
What is the number of probes ? Does the recorder should know about the probes ?

irimiab, 2018-09-03:
I wrote above, in the specs:

> The recorder must be configured with the number of probes to query. So in `httprec.json` we should probably have these settings for the traffic thing:
>
>- url
>- username
>- password
>- number of probes

sparvu, 2018-10-15:
will be available in 1.4

sparvu, 2018-12-07:
u msg is very ok. we will build on top of it. will simulate you data in a bit.

sparvu, 2018-12-07:
merging my latest changes for floud data messages. objects must comply too, not all cameras can install floud. therefore the object must change

irimiab, 2018-12-07:
Did you edit the message? Or what changes?

sparvu, 2018-12-07:
just wait a bit. writing now

sparvu, 2018-12-07:
Details:

 * DS Object name: axis_q1615emk2 (we are talking abut this hdw only here)

 * Messages:

    * floud 

    * id: aqd-q1615emk2-floud

    * raw data: q1615emk2.floud.krd

Details below:

```
      "axis_q1615emk2": {
          "description": "AXIS Q1615-E Mk II Network Camera",
          "messages": {
            "floud": {
              "id": "aqd-q1615emk2-floud",
              "rawdata": "aqd_floud.krd",
              "stall": 120,
              "sli": "yes",
              "format": {
                "fields_separator": ":",
                "fields": [
                  {
                    "name": "subscription_id",
                    "type": "metadata",
                    "role": "subscription_id"
                  },
                  {
                    "name": "ds_id",
                    "type": "metadata",
                    "role": "ds_id"
                  },
                  {
                    "name": "timestamp",
                    "type": "timestamp",
                    "timestamp_format": "unix"
                  },
                  {
                    "name": "probe_id",
                    "type": "metadata",
                    "role": "device_id"
                  },
                  {
                    "name": "count_cls0",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls0",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls0",
                    "type": "numeric"
                  },
                  {
                    "name": "count_cls1",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls1",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls1",
                    "type": "numeric"
                  },
                  {
                    "name": "count_cls2",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls2",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls2",
                    "type": "numeric"
                  },
                  {
                    "name": "count_cls3",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls3",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls3",
                    "type": "numeric"
                  },
                  {
                    "name": "count_cls4",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls4",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls4",
                    "type": "numeric"
                  },
                  {
                    "name": "count_cls5",
                    "type": "numeric"
                  },
                  {
                    "name": "avglen_cls5",
                    "type": "numeric"
                  },
                  {
                    "name": "avgspeed_cls5",
                    "type": "numeric"
                  },
                  {
                    "name": "checksum",
                    "type": "hash",
                    "function": "sha256"
                  }
                ]
              },
              "statistics": [
              ]
            }
          }
      },
```

sparvu, 2018-12-07:
This is the complete object definition.

sparvu, 2018-12-07:
this is a very simple dummy probe:

```
1544190529:probe0:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190529:probe1:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190529:probe2:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190529:probe3:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190529:probe4:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190534:probe0:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190534:probe1:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190534:probe2:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190534:probe3:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190534:probe4:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190539:probe0:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190539:probe1:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190539:probe2:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190539:probe3:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
1544190539:probe4:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:0.00:2.00:2.00:432.00:1.00:456.00:7.00:686.28:5.00:1327.40
```

4 probes, 6 classes.

sparvu, 2018-12-07:
im enabling now the transporter now.

sparvu, 2018-12-16:
Latest update:

  * it is highly more simple to write a basic data recorder to fetch data, and display it. The integration with httprec it is complicated (configuration, perl modules)

  * in order to provide quick access to camera traffic data, we will use this new data recorder which will later be integrated to httprec 

  * the new recorder is able to fetch traffic data from network camera, remains 2 bugs to be fixed in processing the probes and thats it

sparvu, 2018-12-17:
Latest:

 * implemented a simple recorder, called axisrec 

 * works like this

```
% ./axisrec 
1545023479:0:2:3562.00:79.50:0:0:0:0:0:0:0:0:0:5:1758.80:66.20:0:0:0

% ./axisrec -p2
1545023482:0:2:3562.00:79.50:0:0:0:0:0:0:0:0:0:1:630.00:43.00:0:0:0
1545023483:1:0:0:0:1:174.00:6.00:1:436.00:42.00:4:545.50:32.25:15:2009.33:47.80:0:0:0
```

 * usage:
```
% ./axisrec -h
USAGE: axisrec [-hlvV] [-p number] | [interval]
OPTIONS:
  -h        : help information
  -l        : log raw data to krd file, no STDOUT
  -p number : number of probes, default 1
  -v        : verbose information
  -V        : release version
  interval  : maximum number of seconds between samples

 FIELDS:

 AXIS FLOUD MESSAGE (axis_q1615emk2)
  #01 timestamp  : seconds since Epoch, time
  #02 probe      : probe number, number
  #03 count c0   : count class0, number
  #04 avglen c0  : average length class0, number
  #05 avgspd c0  : average speed class0, number
  #06 count c1   : count class1, number
  #07 avglen c1  : average length class1, number
  #08 avgspd c1  : average speed class1, number
  #09 count c2   : count class2, number
  #10 avglen c2  : average length class2, number
  #11 avgspd c2  : average speed class2, number
  #12 count c3   : count class3, number
  #13 avglen c3  : average length class3, number
  #14 avgspd c3  : average speed class3, number
  #15 count c4   : count class4, number
  #16 avglen c4  : average length class4, number
  #17 avgspd c4  : average speed class4, number
  #18 count c5   : count class5, number
  #19 avglen c5  : average length class5, number
  #20 avgspd c5  : average speed class5, number
```

sparvu, 2018-12-17:
* you have real data now, u env same subscription

* check air quality data, you have that missing 

* review with @DTPopa for his demo

No other actions.

irimiab, 2018-12-17:
The traffic data stopped at 7:47 today. I thought you are working on that.

DTPopa, 2018-12-17:
I think there is something wrong I see files for non-existent probes full of phoney data. "q1615emk2.floud.probe**4**.2018-12-17.krd"

irimiab, 2018-12-17:
These raw data files maybe were from the mockup data received before. But I reset the stats and removed the unused devices.

sparvu, 2018-12-17:
Working or not ?

irimiab, 2018-12-17:
Working, yes

DTPopa, 2018-12-17:
Working, but not ok. The data sent to Kronometrix is different from the data obtained via the application's API. therefore it is something wrong either on the recorder/parser side or on the server side.
'Class 5 issue': the class 5 always gives null values for the traffic while the camera answers with some reasonable, non-zero values:
"cls": 5,
**"count()": 8,**
"min(len)": 1104,
"avg(len)": 2244.25,
"max(len)": 4079,
"min(spd)": 6,
"avg(spd)": 15.375,
"max(spd)": 45

sparvu, 2018-12-17:
fixed. will close this one now. later will open a new CR to think for its final destination: httpec, axisrec etc

DTPopa, 2018-12-17:
NOT fixed.

Did you test this?

Still got NA in the interface and non-zero on the camera:

{
	"id": 6480797,
	"build_version": "v3.4.13",
	"cameraId": "accc8e62e39d",
	"session_id": "accc8e62e39d:20180620:204305",
	"gps_lat": "43.788286",
	"gps_lon": "11.226857",
	"query": "getStatsPerClass",
	"localtime": "2018-12-17T18:47:45",
	"interval": 60,
	"probe": 0,
	"count": [
		{
			"cls": 0,
			"count()": 1,
			"min(len)": 32902,
			"avg(len)": 32902.0,
			"max(len)": 32902,
			"min(spd)": 274,
			"avg(spd)": 274.0,
			"max(spd)": 274
		},
		{
			"cls": 4,
			"count()": 1,
			"min(len)": 664,
			"avg(len)": 664.0,
			"max(len)": 664,
			"min(spd)": 4,
			"avg(spd)": 4.0,
			"max(spd)": 4
		},
		{
			"cls": 5,
			"count()": 3,
			"min(len)": 3249,
			"avg(len)": 13503.3333333333,
			"max(len)": 25077,
			"min(spd)": 20,
			"avg(spd)": 121.666666666667,
			"max(spd)": 281
		}
	]
}

DTPopa, 2018-12-17:
Messages are not coming on regular intervals, thus loosing the 1-minute data.
For example messages came 20:51, 20:54, 20:57, 20:59...
[image]

sparvu, 2018-12-17:
* NA: nope nothing here. axisrec does not report NA at this moment

 * I think this is on the app side, @irimiab : can u have a look

This is for example the latest raw data from recorder:

```
1545075977:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545075977:probe1:0:0:0:0:0:0:0:0:0:1:554.00:47.00:2:691.00:67.50:0:3:1083.67:51.67
1545076037:probe0:2:10068.00:2.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076037:probe1:0:0:0:0:0:0:0:0:0:1:572.00:35.00:1:784.00:67.00:0:5:1576.40:64.20
1545076097:probe0:1:71551.00:17.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076097:probe1:0:0:0:0:0:0:0:0:0:0:0:0:1:748.00:56.00:0:3:876.67:59.33
1545076157:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076157:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:2:2113.00:101.00
1545076217:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076217:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:4:1122.50:70.50
1545076277:probe0:1:28359.00:33.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076277:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:1467.33:74.00
1545076337:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:1:2587.00:1.00
1545076337:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076397:probe0:2:32915.50:18.50:0:0:0:0:0:0:0:0:0:0:0:0:0:1:8930.00:4.00
1545076397:probe1:1:832.00:43.00:0:0:0:0:0:0:1:542.00:16.00:0:0:0:0:6:2640.50:50.33
1545076457:probe0:1:9763.00:201.00:0:0:0:0:0:0:0:0:0:0:0:0:0:2:12742.00:100.50
1545076457:probe1:0:0:0:0:0:0:0:0:0:1:600.00:22.00:2:673.00:66.50:0:3:5427.33:113.00
1545076517:probe0:1:19568.00:39.00:0:0:0:0:0:0:0:0:0:0:0:0:0:2:2424.00:5.50
1545076517:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076577:probe0:1:29549.00:30.00:0:0:0:0:0:0:0:0:0:0:0:0:0:1:3600.00:4.00
1545076577:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:4:5488.50:47.75
1545076637:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076637:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:6:3451.00:31.17
1545076697:probe0:1:37962.00:23.00:0:0:0:0:0:0:0:0:0:0:0:0:0:1:4426.00:2.00
1545076697:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:4:2905.50:36.75
1545076757:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:2:18387.00:8.50
1545076757:probe1:0:0:0:0:0:0:0:0:0:1:528.00:2.00:0:0:0:0:5:6500.40:33.80
1545076817:probe0:2:11199.50:4.00:0:0:0:0:0:0:0:0:0:0:0:0:0:4:14073.00:5.00
1545076817:probe1:0:0:0:0:0:0:0:0:0:2:508.00:26.00:0:0:0:0:3:4119.33:61.67
1545076877:probe0:2:35725.00:216.00:0:0:0:0:0:0:0:0:0:0:0:0:0:1:17962.00:105.00
1545076877:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:4:1831.25:58.00
1545076937:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076937:probe1:0:0:0:0:0:0:1:450.00:61.00:0:0:0:1:686.00:58.00:0:4:1220.00:68.00
1545076997:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545076997:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:7:1764.29:71.29
1545077057:probe0:1:48591.00:85.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077057:probe1:0:0:0:0:0:0:0:0:0:0:0:0:1:684.00:42.00:0:6:1219.00:54.50
1545077117:probe0:2:30528.00:32.50:0:0:0:0:0:0:0:0:0:0:0:0:0:1:2436.00:2.00
1545077117:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:5:1916.00:78.00
1545077177:probe0:1:35772.00:126.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077177:probe1:0:0:0:0:0:0:1:410.00:56.00:0:0:0:0:0:0:0:3:1133.33:61.67
1545077237:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077237:probe1:0:0:0:0:0:0:0:0:0:2:509.00:46.00:1:664.00:45.00:0:3:1148.67:52.33
1545077297:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077297:probe1:0:0:0:0:0:0:1:440.00:42.00:0:0:0:1:654.00:25.00:0:0:0
1545077357:probe0:4:16532.50:11.25:0:0:0:0:0:0:0:0:0:0:0:0:0:3:6328.00:3.67
1545077357:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:5:2338.80:58.40
1545077417:probe0:1:67250.00:228.00:0:0:0:0:0:0:1:519.00:7.00:0:0:0:0:3:2720.67:45.00
1545077417:probe1:0:0:0:0:0:0:1:438.00:0.00:0:0:0:0:0:0:0:1:4248.00:10.00
1545077477:probe0:1:16230.00:32.00:0:0:0:0:0:0:0:0:0:0:0:0:0:4:12915.00:73.25
1545077477:probe1:1:956.00:326.00:0:0:0:0:0:0:0:0:0:0:0:0:0:4:1825.25:2.75
1545077537:probe0:2:23097.50:213.50:0:0:0:0:0:0:0:0:0:0:0:0:0:1:3112.00:163.00
1545077537:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:12:3357.08:13.92
1545077597:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:1:2313.00:7.00
1545077597:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:2:5857.50:8.00
1545077657:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077657:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:4400.33:5.00
1545077717:probe0:2:24166.00:184.00:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077717:probe1:0:0:0:0:0:0:1:306.00:0.00:0:0:0:0:0:0:0:4:3705.50:3.00
1545077777:probe0:1:50676.00:300.00:0:0:0:0:0:0:0:0:0:0:0:0:0:5:5443.00:54.20
1545077777:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545077837:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:9790.33:28.00
[... 53 lines cut ...]
1545079457:probe0:1:6818.00:36.00:0:0:0:0:0:0:0:0:0:0:0:0:0:4:7692.25:73.50
1545079457:probe1:0:0:0:0:0:0:0:0:0:1:478.00:46.00:1:724.00:49.00:0:2:1025.00:58.50
1545079517:probe0:1:44985.00:79.00:0:0:0:0:0:0:0:0:0:0:0:0:0:2:3267.50:4.50
1545079521:probe1:0:0:0:0:0:0:0:0:0:0:0:0:2:661.00:47.00:0:3:1537.33:57.33
1545079577:probe0:1:20788.00:268.00:0:0:0:0:0:0:0:0:0:0:0:0:0:3:3026.00:44.67
1545079577:probe1:0:0:0:0:0:0:0:0:0:0:0:0:1:642.00:54.00:0:1:1140.00:51.00
1545079637:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545079638:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
1545079697:probe0:0:0:0:0:0:0:0:0:0:0:0:0:1:718.00:97.00:0:5:11705.00:59.20
1545079697:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:1:1114.00:50.00
1545079757:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:2:2370.00:104.50
1545079758:probe1:0:0:0:0:0:0:0:0:0:0:0:0:1:678.00:41.00:0:4:1825.00:48.50
1545079817:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
[... cut ...]
1545079937:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:4:1998.75:74.00
1545079937:probe1:0:0:0:0:0:0:0:0:0:0:0:0:3:710.00:59.67:0:0:0
1545079997:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:1:3286.00:82.00
1545079998:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:1068.33:66.33
1545080057:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:2251.00:114.00
1545080057:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:1:1896.00:71.00
1545080117:probe0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:3:1185.33:80.67
1545080118:probe1:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
```

sparvu, 2018-12-17:
axisrec it is not integrated yet as I have explained. It runs experimental on some dev machine inside kronometrix.com - you can move it to your own network as you want or wait before we are ready with kronometrix.io and install it final on production.

Time was wrong. Fixed. Let me know if there is still a problem.

sparvu, 2018-12-17:
i have invited myself using bogdan id to the AQ subscription. Please add users for future. 
Now I see the data ok.

sparvu, 2018-12-17:
And now i cannot see anymore anything. wtf again this vpn probable. let me know if you can see or not data when u can. 

[image]

DTPopa, 2018-12-17:
What is this? I don;t understand anything from these plots which are not continuous.
[image]

DTPopa, 2018-12-17:
Sorry boys, I don't even have something to test here. Let me know when I can use something.

sparvu, 2018-12-17:
aha. again 

 * first of all something is wrong again with the VPN. I hope your net is working 

 * let me check again the axisrec, it was running fine today. right ? i dont know what is this

sparvu, 2018-12-17:
first of all, when u can reboot your router wan.

sparvu, 2018-12-17:
i have checked the recorder. restarted and the transporter as well.

sparvu, 2018-12-17:
ok, it was a damn typo after i fixed class 5. now it should be just fine. always check availability app to see the real problem. in this case the avail app was showing something wrong with the transmitter, sending part.

sparvu, 2018-12-18:
so clarify pls if this is ok or not

sparvu, 2018-12-18:
re-opened.

DTPopa, 2018-12-18:
Meeting. Nu mai umbla la nimic.

irimiab, 2018-12-18:
I would propose to log the JSON read from the camera for a period of time. For example, one file per minute, containing the JSON data returned by the camera when interrogated. We will compare them with the actual data received in Kronometrix and eliminate any suspicions about the recorder. 

If possible, do this without interrupting the current data flow.

What do you think?

sparvu, 2018-12-18:
here it is:  1st is the json and followed up by the raw krd formatted data

```
$ cat aqd_q1615emk2_floudrec.krd 
{
	"id": 6480797,
	"build_version": "v3.4.13",
	"cameraId": "accc8e62e39d",
	"session_id": "accc8e62e39d:20180620:204305",
	"gps_lat": "43.788286",
	"gps_lon": "11.226857",
	"query": "getStatsPerClass",
	"localtime": "2018-12-18T15:03:36",
	"interval": 60,
	"probe": 0,
	"count": [
		{
			"cls": 0,
			"count()": 5,
			"min(len)": 2334,
			"avg(len)": 11505.8,
			"max(len)": 24925,
			"min(spd)": 10,
			"avg(spd)": 77.0,
			"max(spd)": 139
		},
		{
			"cls": 4,
			"count()": 1,
			"min(len)": 628,
			"avg(len)": 628.0,
			"max(len)": 628,
			"min(spd)": 60,
			"avg(spd)": 60.0,
			"max(spd)": 60
		},
		{
			"cls": 5,
			"count()": 5,
			"min(len)": 1239,
			"avg(len)": 2687.4,
			"max(len)": 3633,
			"min(spd)": 19,
			"avg(spd)": 71.0,
			"max(spd)": 128
		}
	]
}
1545145417:probe0:5:11505.80:77.00:0:0:0:0:0:0:0:0:0:1:628.00:60.00:5:2687.40:71.00
{
	"id": 6480797,
	"build_version": "v3.4.13",
	"cameraId": "accc8e62e39d",
	"session_id": "accc8e62e39d:20180620:204305",
	"gps_lat": "43.788286",
	"gps_lon": "11.226857",
	"query": "getStatsPerClass",
	"localtime": "2018-12-18T15:03:36",
	"interval": 60,
	"probe": 1,
	"count": [
		{
			"cls": 0,
			"count()": 2,
			"min(len)": 3788,
			"avg(len)": 8176.5,
			"max(len)": 12565,
			"min(spd)": 80,
			"avg(spd)": 96.0,
			"max(spd)": 112
		},
		{
			"cls": 2,
			"count()": 1,
			"min(len)": 412,
			"avg(len)": 412.0,
			"max(len)": 412,
			"min(spd)": 30,
			"avg(spd)": 30.0,
			"max(spd)": 30
		},
		{
			"cls": 3,
			"count()": 2,
			"min(len)": 460,
			"avg(len)": 502.0,
			"max(len)": 544,
			"min(spd)": 30,
			"avg(spd)": 30.5,
			"max(spd)": 31
		},
		{
			"cls": 4,
			"count()": 3,
			"min(len)": 620,
			"avg(len)": 685.333333333333,
			"max(len)": 726,
			"min(spd)": 32,
			"avg(spd)": 54.0,
			"max(spd)": 70
		},
		{
			"cls": 5,
			"count()": 8,
			"min(len)": 952,
			"avg(len)": 2677.75,
			"max(len)": 6286,
			"min(spd)": 35,
			"avg(spd)": 80.875,
			"max(spd)": 158
		}
	]
}
1545145417:probe1:2:8176.50:96.00:0:0:0:1:412.00:30.00:2:502.00:30.50:3:685.33:54.00:8:2677.75:80.88
```

irimiab, 2018-12-18:
That's good; we will analyze the data for an hour or so and make sure that whatever numbers we have in Kronometrix is exactly what the camera returns.

sparvu, 2018-12-18:
here. 30mins of data.

[axis.floud.debug.tar.gz](https://github.com/kronometrix/recording/files/2691223/axis.floud.debug.tar.gz)

sparvu, 2018-12-18:
so, what is the result ? good / bad / other ?

DTPopa, 2018-12-18:
The aim of the testing is not to assess the correspondence between Json and output of the recorder. That is your responsibility.
The aim is to assess the system functionality as a whole and, if there are deviations, to find the root cause.

sparvu, 2018-12-18:
The current CR is reporting certain defects in the axisrec. Let me know if the recorder is ok or not and if data which arrives to you is ok. That was clearly the aim at 5pm today. 

For the rest: overall testing, system integration thats a different discussion.

DTPopa, 2018-12-18:
Care e requestul pe care il transmite recorderul si la ce interval?

sparvu, 2018-12-18:
I think these were the specs defined for it:

For each probe, the recorder should do a request to this URL (with basic auth):
```
http://<url>/local/VehicleCounter/getStatsPerClass.cgi?interval=60&pid=<probeID>
```
where `<url>` is the URL of the camera and `<probeID>` is a number from 0 to `number_of_probes-1` (probes are 0-based indexed). The recorder should run **every minute** because the parameter `interval=60` in the request URL means we are querying for data for the last 60 seconds.

DTPopa, 2018-12-18:
1. After the reset we get NA all over, therefore something is wrong.
2. For future add "&once=1" in the end.

[12 more replies]

closed 2018-12-19 by sparvu